mantidproject/mantid #35235: Windows Clean/Dirty Install Smoke Tests  :soap: :hankey:

sf1919, 2023-02-17:
We must test both the .exe installer package and install direct from conda.

* To install via conda:
  - Use Intel Conda and make sure conda-forge is added to channels
```
mamba create -n test_mantid_6.6
mamba activate test_mantid_6.6
mamba install -c mantid/label/rc1 mantidworkbench --yes
workbench # to launch
```

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :soap: :hankey:

### Dirty install

* Make sure that you have several versions of Mantid installed
 * Last release
 * A nightly
 * If possible an old release
* Install the new release candidate of Mantid
- [ ] Check that Mantid boots up correctly

### Clean install

* Remove all existing Mantid versions and associated files

**On Windows**:
 * Uninstall the program
 * Clear shortcuts from desktop
 * Remove the mantid  files in %APPDATA%

**On macOS** :
 * Remove the application
 * Remove the `~/.mantid directory`
 * Remove (or rename) `~/Library/Preferences/org.python.python.Python.plist`  and `~/Library/Preferences/org.python.python.plist` 

**On Linux** :
 * Remove the package: `/opt/Mantid`
 * Remove `~/.config/Mantid`
 * Remove `~/.mantid/`

Re-install the new release candidate of Mantid

- [ ] Check that Mantid boots up correctly

MohamedAlmaki, 2023-02-17:
When I tried to run the workbench directly after installing it using mamba I got "Cannot find qt plugin in windows platform" but when closing and reopening the PowerShell the workbench works fine

sf1919, 2023-02-17:
The original problem was recorded in #34923 and fixed. This seems to be a slightly new way of getting this error but there does appear to be a workaround

closed 2023-02-17 by sf1919