mantidproject/mantid #35252: Linux Plotting Smoke Tests  :chart_with_upwards_trend: :bar_chart:

sf1919, 2023-02-17:
* To install the tar.xz package for Linux, run `(sudo) tar -xJf mantid-VA.B.C.tar.xz` in a terminal and it will unzip the package in your current working directory. 
* To install via conda:
  - Use Intel Conda and make sure conda-forge is added to channels
```
mamba create -n test_mantid_6.6
mamba activate test_mantid_6.6
mamba install -c mantid/label/rc1 mantidworkbench --yes
workbench # to launch
```

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :chart_with_upwards_trend: :bar_chart:

* 1D plotting:[instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/03_displaying_1D_data.html#displaying-1d-data)
 - [ ] Simple plot 
 - [ ] Another way to plot 
 - [ ] Adding curves to existing plots 
 - [ ] Also, test out [waterfall](https://docs.mantidproject.org/nightly/plotting/WaterfallPlotsHelp.html#waterfall-plots) and [tiled]( https://docs.mantidproject.org/nightly/plotting/1DPlotsHelp.html#tiled-plots)
 - [ ] Check Toolbar buttons

* 2D plotting: [instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/04_displaying_2D_data.html#displaying-2d-data)
 - [ ] Plot all spectra 
 - [ ] Change colour map 
 - [ ] [Contour plot](https://docs.mantidproject.org/nightly/plotting/ColorfillPlotsHelp.html#contour-plots) (under 3D menu) 
 - [ ] Check Toolbar buttons

* [3D plotting](https://docs.mantidproject.org/nightly/plotting/3DPlotsHelp.html):
 - Load some data eg `LOQ7041` from the ISIS sample data
 - [ ] 3D surface
 - [ ] 3D wire frame  
 - [ ] Check Toolbar buttons

 ## Sliceviewer
 - [ ] Overly long instructions (don't spend  long!) and data [here](https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html). In particular try editing the data in a workspace while it is open in Sliceviewer!

darominski, 2023-02-17:
1D plotting:
1. When modifying parameters of an waterfall plot, and both offsets are set to zero, the following exception appears:
File "lib/python3.8/site-packages/mantidqt/widgets/waterfallplotoffsetdialog/presenter.py", line 27, in <lambda>
    self.view.x_offset_spin_box.valueChanged.connect(lambda: self.update_x_offset())
  File "lib/python3.8/site-packages/mantidqt/widgets/waterfallplotoffsetdialog/presenter.py", line 42, in update_x_offset
    self.ax.update_waterfall(new_x, self.view.get_y_offset())
  File "lib/python3.8/site-packages/mantid/plots/mantidaxes.py", line 1128, in update_waterfall
    self.set_waterfall_fill(False)
  File "envs/test_mantid_6.6/lib/python3.8/site-packages/mantid/plots/mantidaxes.py", line 1209, in set_waterfall_fill
    raise RuntimeError("Cannot toggle fill on non-waterfall plot.")
RuntimeError: Cannot toggle fill on non-waterfall plot.

darominski, 2023-02-17:
[image]

darominski, 2023-02-17:
Issues opened for 1D plotting issue (#35257 ) and the sliceviewer (#35260). The 2D plotting issue is minor.

closed 2023-02-17 by darominski