mantidproject/mantid #35258: Multiples of Single GUIs on Project Loading

ConorMFinn, 2023-02-17:
**Describe the bug**
When loading a project, interfaces like the Reflectometry GUI can have multiple versions of themselves open, which should not be possible. 

**To Reproduce**
1. Open the refl GUI
2. Save the project
3. Close workbench
4. Open and reload the project
5. Find the (hopefully) opened Refl GUI and try to open another one
6. You should now have two. 

**Expected behavior**
Only one GUI should be visible and intractable. 

**Screenshots**
[image]

**Platform/Version (please complete the following information):**
 - OS: All
 - Mantid Version At least 6.5

**Additional context**

ConorMFinn, 2023-10-25:
It is, removed.

rbauststfc, 2024-01-15:
For LSS I can only replicate this with the Reflectometry GUI, so I'm going to label it as a Reflectometry issue.

closed 2026-03-03 by github-actions[bot]