mantidproject/mantid #35260: Sliceviewer: add peaks overlay changes view layout

darominski, 2023-02-17:
**Describe the bug**

There is an issue with Sliceviewer where if the Sliceviewer window is not resized and `Add peaks overlay` is selected and then approved (pressed `OK`), the plot in the middle is permanently resized to occupy only a portion of the window, as shown in the screenshot below.

**To Reproduce**
1. Follow the instruction of the Sliceviewer testing [guide](https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html) or run the following lines to create some 4D data:

```
from mantid.simpleapi import *

md_4D = CreateMDWorkspace(Dimensions=4, Extents=[0,2,-1,1,-1.5,1.5,-0.25,0.25], Names="H,K,L,E", Frames='HKL,HKL,HKL,General Frame',Units='r.l.u.,r.l.u.,r.l.u.,meV')
FakeMDEventData(InputWorkspace=md_4D, UniformParams='5e5') # 4D data
FakeMDEventData(InputWorkspace=md_4D, UniformParams='1e5,0.5,1.5,-0.5,0.5,-0.75,0.75,-0.1,0.1') # 4D data

# Add a non-orthogonal UB
expt_info = CreateSampleWorkspace()
md_4D.addExperimentInfo(expt_info)
SetUB(Workspace='md_4D', c=2, gamma=120)
```
2. Select `md_4D` and visualize the data in the sliceviewer
3. Select `Add peaks overlay on/off` and press `OK` in the dialog.
4. Observe that the centre plot occupies only a portion of the available space. 

**Expected behavior**

The view should not permanently resize.

**Screenshots**
[image]

**Platform/Version (please complete the following information):**
 - OS: Ubuntu
 - OS Version: 20.04
 - Mantid Version 6.6.0.rc1

**Additional context**

gemmaguest, 2023-02-17:
Reproduced in v6.4.0 Ubuntu

jhaigh0, 2023-09-06:
Still and issue

closed 2023-10-18 by peterfpeterson