mantidproject/mantid #35261: Engineering Diffraction Interface hanging upon opening after indexing data archive

MialLewis, 2023-02-17:
Related to https://github.com/mantidproject/mantid/issues/34820.

After browsing to the data archive, the same hang occurs when trying to open the Engineering Diffraction interface.

To improve ease of use, it could be worth reworking whatever is hang to reoccur in the interface.

**To Reproduce**

1.  Follow steps in https://github.com/mantidproject/mantid/issues/34820
2. Try to open the Engineering Diffraction interface

**Expected behavior**
No hang when opening the Eng Diff interface

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

thomashampson, 2023-05-16:
Was this caused by the non-native browser? It should be fixed now if that's the case.

jclarkeSTFC, 2024-01-16:
Just tried this in a dev build on IDAaaS, the Engineering Diffraction interface hangs on opening, can't close it or do anything else with it apart from killing Workbench.