mantidproject/mantid #35301: Support loading of data files with non-ASCII characters in their path

thomashampson, 2023-02-23:
**Original reporter:** Gregory Smith @ ISIS

**Describe the bug**
When trying to reduce ISIS SANS2D data in the ISIS SANS Reduction GUI, reduction was not successful on Windows. We think this was because the file path contained non-ASCII characters. When using the file loading dialog it lets you know that it can't parse the file name, but the SANS reduction doesn't give any useful information. It would be good if we could just load these files.

**To Reproduce**
1. Move a .nxs file into a directory containing a non-ascii character, e.g. ü.
2. Try to load the file.

**Expected behavior**
File will load as normal.

thomashampson, 2023-04-20:
@rbauststfc I think it's quite a big code change to support this in general. Would it be possible in the meantime to give the user some feedback in the SANS GUI when this happens?

rbauststfc, 2023-04-21:
Thanks for looking into it @thomashampson, we'll add this issue to our SANS to-do list then. Jonathan confirmed that you're not planning to keep an issue open for the underlying fix, so we'll close this issue once we've made our change.

closed 2023-10-27 by github-actions[bot]