mantidproject/mantid #35367: Sefault when importing mantid simple api in github workflow

MialLewis, 2023-03-15:
**Describe the bug**
I've been trying to build a package with `mantid` as a dependency. On push, a github workflow runs unit tests that import `simpleapi`. This causes a segfault. Example: https://github.com/MialLewis/mantidSetup/actions/runs/4424802039/jobs/7759064055

I have distilled the package down to the barebones, but the segfault still persists: https://github.com/MialLewis/mantidSetup

This segfault does not occur if changing the `Test Mantid Import` statement here to `python -c "from mantid import kernel"` etc., so I am inferring that it is related to `simpleapi`.

Importantly, if you import `simpleapi` at the start of a script, it will run the whole script successfully (all unit tests in this case) and then segfault when the script end/python session terminates.

I have a core dump which doesn't seem to be much help, too large to upload here so message if you want it. Human readable part here:
```
ProblemType: Crash
Architecture: amd64
Date: Tue Mar 14 14:28:09 2023
DistroRelease: Ubuntu 22.04
ExecutablePath: /usr/share/miniconda/envs/test-env/bin/python3.8
ExecutableTimestamp: 1678803712
ProcCmdline: python -c from\ mantid\ import\ simpleapi
ProcCwd: /home/runner/work/mantidSetup/mantidSetup
ProcEnviron:
 LANG=C.UTF-8
 XDG_RUNTIME_DIR=<set>
 PATH=(custom, user)
```

SilkeSchomann, 2023-03-15:
When adding `python -c "from mantid import simpleapi"` to the MSlice unit test workflow there is no segfault but the workflow starts to use Mantid 6.5 instead of Mantid nightly without telling us why: https://github.com/mantidproject/mslice/actions/runs/4405918108/jobs/7717444682

MialLewis, 2023-03-16:
To provide an update, I have narrowed this down to the downloading of instrument data when `FrameworkManagerImpl.Instance()` is ran.

You can replicate this on ubuntu (other OS' not investigated) by first removing the mantid cache to enable the downloading of the definitions (`rm -r ~/.mantid`), then running the following script:
```
from mantid.api import FrameworkManagerImpl as FMI
FMI.Instance()
```

jhaigh0, 2023-05-11:
Just wanted to check I'm testing this right...

I'm on Ubuntu, I've made a new conda environment called `mantid-test`, I've installed `mantid` into it, then I've done
```
(mantid-test) jonathan@jonathan-VirtualBox:~$ rm -r ~/.mantid/
(mantid-test) jonathan@jonathan-VirtualBox:~$ python
Python 3.8.15 | packaged by conda-forge | (default, Nov 22 2022, 08:46:39) 
[GCC 10.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from mantid.api import FrameworkManagerImpl as FMI
>>> FMI.Instance()
FrameworkManager-[Notice] Welcome to Mantid 6.6.0
FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid6.6
Python-[Warning] plotSpectrum not available
DownloadInstrument-[Notice] Downloading 15 files from the instrument repository
<mantid.api._api.FrameworkManagerImpl object at 0x7fa5f9200e40>
>>> DownloadInstrument-[Notice] A new Facilities.xml file has been downloaded, this will take effect next time Mantid is started.
>>> 
```

So it resulted in no crash for me. My developer build is broken on Ubuntu, so I couldn't test there. Do you know there the cache is on Windows, so I can have a go with a dev build?

MialLewis, 2023-05-11:
You need to use `python -c <command here>` rather than calling opening python then entering your commands. The issue occurs when the python instance (thread?) is destroyed before the instrument data has finished downloading.

jhaigh0, 2023-05-11:
> You need to use `python -c <command here>` rather than opening python then entering your commands. The issue occurs when the python instance (thread?) is destroyed before the instrument data has finished downloading.

Ah yes got that to segfault now thanks.

jhaigh0, 2023-05-11:
> > You need to use `python -c <command here>` rather than opening python then entering your commands. The issue occurs when the python instance (thread?) is destroyed before the instrument data has finished downloading.
> 
> Ah yes got that to segfault now thanks.

Interestingly, it seems to work fine on Windows

jhaigh0, 2023-05-11:
So I think the [long comment](https://github.com/mantidproject/mantid/blob/main/qt/applications/workbench/workbench/app/start.py#L250) at the bottom of `start.py` might be important...
```
# Mantid's FrameworkManagerImpl::Instance Python export uses a process-wide static flag to ensure code
# only runs once (see Instance function in Framework/PythonInterface/mantid/api/src/Exports/FrameworkManager.cpp).
# The default start method on Unix ('fork') inherits resources such as these flags from the parent.
# We require the start method to be 'spawn' so that we do not inherit these resources from the parent process,
# this is already the default on Windows/macOS.
# This will mean the relevant 'atexit' code will execute in the child process, and therefore the
# FrameworkManager and UsageService will be shutdown as expected.
```
So it sounds like how Unix deals with threads by default is not what we want.
Reading about the python `multiprocessing` module, it says _Note that safely forking a multithreaded process is problematic._
This might explain why it works fine on windows.

So I tried adding in `set_start_method('spawn')` to the test script on Ubuntu and I got a connection error rather than a segfault, which is promising.
```
(mantid-test) jonathan@jonathan-VirtualBox:~$ python -c "import multiprocessing as mp; mp.set_start_method('spawn'); from mantid.api import FrameworkManagerImpl as FMI; FMI.Instance()"
FrameworkManager-[Notice] Welcome to Mantid 6.6.0
FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid6.6
DownloadInstrument-[Notice] Internet Connection Failed - cannot update instrument definitions. Please check your connection. If you are behind a proxy server, consider setting proxy.host and proxy.port in the Mantid properties file or using the config object.
```
So that might be a bit of a workaround to try

jhaigh0, 2023-07-19:
Been struggling to recreate the problem now. To check it's a threading thing, do you think you could run `taskset 1 python -c ...` to force python onto one core? Unsure if it'll even let you do that, but would be interesting to see.

A comment [here](https://unix.stackexchange.com/questions/467678/run-process-as-if-on-a-single-core-machine-to-find-a-bug) had luck debugging a threading problem using `strace` so maybe running `strace python -c ...` would yield some results.

jhaigh0, 2023-11-14:
Was this fixed by https://github.com/mantidproject/mantid/pull/36273?

MialLewis, 2023-11-14:
Yep this is fixed