mantidproject/mantid #35373: Add support for a gauge volume in DiscusMultipleScatteringCorrection

DannyHindson, 2023-03-17:
I had a discussion with Joe Kelleher about using the DiscusMultipleScatteringCorrection on the ENGINX instrument. They have a radial collimator that restricts single scatter tracks to scatter points within a small region within the larger sample volume. Double scatter tracks however may manage to pass through the collimator even if the two scatter points are outside the gauge volume eg a track could undergo two 45 degree scatters and still arrive at the ENGINX banks which are positioned at approximately 90 degrees to the beam

This issue is to add support for a collimator to the Discus algorithm. Here are a couple of possible approaches:

a) leave the algorithm "as is" and set up the collimator as a sample environment component that attenuates tracks that don't pass through the collimator slits
b) modify the algorithm to force the final scatter point on any tracks to happen in a part of the sample where the final track segment can pass through the collimator
c) modify the algorithm to assign zero weight to tracks where the final scatter is not in a position that allows the final track segment to pass through the collimator

I had a look into a). The script on issue https://github.com/mantidproject/mantid/pull/35369 sets up something that approximates the ENGINX collimator on one side of the beam by building up a CSG shape consisting of ~160 cuboids. The position of the slits isn't quite right but it allowed me to do a performance test. This showed that the presence of a lot of additional CSG shapes slows down the algorithm a lot. A calculation using just 10 scenarios went from ~8 seconds up to about 650 seconds. This happens because the code tests the outbound track for intersections with every single surface in the collimator CSG object (and there are ~960 surfaces to test vs only 3-6 for the sample depending on the shape). It might be possible to add an optimisation that subselects the active surfaces for a particular detector meaning you only need to consider a couple of the collimator cuboids for a particular detector.

Approach b) is basically an "importance sampling" approach where you don't waste simulation time on neutron paths that won't contribute to the final answer. There is already importance sampling in the Discus algorithm based on S(Q,w) - so it essentially samples theta and w. I think it would be difficult to extend it to handle this collimator case. For each detector there will be a "corridor" through the sample volume where the final scatter point needs to lie. When the code decides on the track direction following the penultimate scatter it would need to select a direction that intersects this "corridor" and that's pretty tricky because you're sampling both theta and phi and doing this kind of importance sampling based on 2 variables is hard.

I think approach c) is probably the best one to try. It's not a million miles away from a) in terms of what's actually happening. It's probably quicker than a) (even if a) was optimised as I suggested) because you don't bother calculating track intersections with the collimator.

DannyHindson, 2023-03-17:
**Implementation approach**

Make a change to the DiscusMultipleScatteringCorrection::scatter() function after the iScat loop at around ~L1434. At this point the final scatter position has been selected. Do the following:

- construct a shape (cylinder?) extending from detPos to the gauge volume
- check if the final scatter point lies inside this shape
- if not assign weight zero. if it does like inside proceed as normal

DannyHindson, 2023-03-17:
Here's a branch with a draft version of the code changes required to implement approach c):

https://github.com/mantidproject/mantid/tree/AddCollimatorSupportToDiscus

RichardWaiteSTFC, 2023-10-31:
Thanks for picking this up @warunawickramasingha 
if you want to run the algorithm from the python side to familiarise yourself with it - you can follow the instructions here
https://github.com/mantidproject/mantid/pull/34762

RichardWaiteSTFC, 2023-11-15:
One suggestion for accepting and event or not relates to determining the angular extent of the collimator opening as seen from the detector (this will depend on azimuthal angle...). Do you think this would be possible?
[image]

RichardWaiteSTFC, 2024-06-17:
Following discussion with Joe K on ENGIN-X - we will require the users to specify the collimator height and it is OK to center the collimator opening on the sample-detector (L2) vector

PR #37648 (2024-09-10): 35373 support for collimator in DiscusMultipleScatteringCorrection algorithm
### Description of work
This work adds support for a radial collimator in `DiscusMultipleScatteringCorrection` algorithm that it restricts scatter points within a small region within the larger sample volume. The algorithm was modified to assign zero weight to tracks where the final scatter is not in a position that allows the final track segment to pass through the collimator toward detectors.

#### Purpose of work
This work allows the use of `DiscusMultipleScatteringCorrection` algorithm for ENGINX instrument that h [... cut]
closed 2024-09-10 by SilkeSchomann