mantidproject/mantid #35435: Mantid freezes when clicking cells in Show Data dialog for specific datasets

rbauststfc, 2023-04-04:
**Original reporter:** Rob from the SANS Group at ISIS

**Describe the bug**

Mantid freezes for specific datasets when a user tries to click a cell after opening the `Show Data` dialog for a loaded workspace. This also seems to vary depending on the computer being used. I have tested with a few different datasets and currently think that the number of histograms rather than simply the size of the dataset may have some significance. I haven't been able to replicate the issue with a dataset like `INTER61170`, which has just 258 histograms, while I have with `INTER74466`, which is actually a smaller dataset than the INTER one with fewer events but has over 40,000 histograms. More histograms means more rows in the data table, so this may have some significance.

In terms of datasets affected, I'm currently aware that this can be replicated with many of the SANS datasets but also with WISH data. I am waiting for our SANS users to confirm how much of an issue this behaviour is currently causing for them.

**To Reproduce**

1) Load a dataset with a large number of histograms. The issue is known to replicate with `LARMOR74465` or `LARMOR74466`.
2) Right click on the workspace in the ADS and select `Show Data`.
3) Click on a cell in the dialog that pops up. After a brief pause, Mantid will freeze and become unresponsive.

**Expected behavior**

Mantid should remain responsive and the clicked cell should appear selected. The expected behaviour should be observable if the above is repeated with a dataset such as `INTER61170`.

**Platform/Version (please complete the following information):**
Further investigation required to identify what impact OS and/or machine spec may be having, but aware of the following so far:
- Has been successfully replicated on my Windows laptop (40GB RAM) on current main.
- Cannot be replicated on IDAaaS in either Mantid version 6.5 or 6.6.
- The original issue was spotted on a machine with 128GB of RAM but an old processor running Mantid version 6.6 - haven't yet confirmed which OS.
- The problem couldn't be replicated on a different desktop machine that was running version 6.5 of Mantid.

smk78, 2023-09-14:
The current workaround to this, as indicated above, is to do the operation on IDAaaS. _However_, it has now been pointed out to me that when IDAaaS removes access to the archive this workaround will no longer be possible (at least not without some considerable pain). I am trying to ascertain the timescale for removal of the archive.

That aside, as indicated above, if the cause of the problem really is the number of histograms then _this issue is not SANS-specific_, it must surely affect any instrument using fine time binning. For that reason I suggest that the 'SANS' and 'ISIS Team: LSS' labels are inappropriate and this should be tasked to the Core team?

MohamedAlmaki, 2023-10-19:
This also happens when I tried to load SXD23767, it looks like a general issue and not just related to LSS?

MohamedAlmaki, 2023-10-19:
I tested this based on a user report. They encounter issues when displaying data for single crystal workspaces. I contacted @jhaigh0, and it works fine for him. However, on my laptop, the interface hangs. So not sure if this is a common problem or not.

rbauststfc, 2023-10-19:
Thanks @MohamedAlmaki, that's useful to know.

jclarkeSTFC, 2023-11-14:
I tried this on my laptop with Windows and Ubuntu 22.04, but couldn't reproduce it.

MohamedAlmaki, 2023-11-15:
@jclarkeSTFC what are your laptop spec?

jclarkeSTFC, 2023-11-15:
> @jclarkeSTFC what are your laptop spec?

i9 @ 2.4 GHz, 8 cores, 64 GB RAM

MohamedAlmaki, 2023-11-15:
mine has 32 GB of RAM and the problem occurs occasionally. I suspect that this issue might be happening on PCs with lower specifications. To reproduce, it could be worth trying to open several applications to increase memory and CPU usage and then attempting to replicate it in Mantid.

jhaigh0, 2023-11-15:
> mine has 32 GB of RAM and the problem occurs occasionally. I suspect that this issue might be happening on PCs with lower specifications. To reproduce, it could be worth trying to open several applications to increase memory and CPU usage and then attempting to replicate it in Mantid.

My laptop had 32GB memory and I can't recreate the issue either. Confusing one.

rbauststfc, 2023-11-15:
Is it worth comparing CPU spec as well as memory? My laptop is i7 2.6GHz, 4 cores

sf1919, 2024-02-02:
Has this been rechecked since the Project Recovery fix went in?

rbauststfc, 2024-02-02:
I was still experiencing the issue yesterday on my laptop when trying to view SANS2D data. My local copy of Mantid is up to date with current main.

MohamedAlmaki, 2024-07-26:
> @MohamedAlmaki are you still able to recreate this?

Yes, I can recreate it using SXD23767, but sometimes it just freezes without a crash