mantidproject/mantid #35438: Update pull request template

rbauststfc, 2023-04-05:
The process of both reviewing and getting gatekeeper reviews for PRs may be helped by ensuring that there is enough information on PRs and in PR review comments. The following is some guidance we could add to a couple of sections of the PR template to ensure developers have the information they need to achieve this:

Under the "Description of work" section:
- [ ] The purpose of the PR should be given.
- [ ] There should be a high level description of the work that was done.
- [ ] If there is no linked issue that provides the context for the PR, then appropriate context should be provided.

Comments for the "Reviewer" section:
- [ ] This section already asks developers to comment on a comprehensive list of items, but we should add a brief explanation of how the comments will be used by gatekeepers so that developers understand why they're being asked to add comments.

gemmaguest, 2023-04-05:
> * There should be sufficient instructions for someone unfamiliar with the application to test - don't assume prior knowledge.

One caveat is that a specialist reviewer may be required in which case some assumptions could be made about prior knowledge; it's not ideal but sometimes is best for expediency. Maybe add something like "(unless a specific reviewer is requested)"?

gemmaguest, 2023-04-05:
> * This section already asks developers to comment on a comprehensive list of items, but we should add a brief explanation of how the comments will be used by gatekeepers so that developers understand why they're being asked to add comments.

Note that the most important comment for gatekeepers is the last one - any changes since a review may invalidate it, so it's worth explaining if anything has changed / why the review is still valid etc. in the final comment.

peterfpeterson, 2023-04-11:
Two comments:
1. There are a fair number of contributors that don't write issues first. Their work is either not scheduled or is stored in a separate system. I like that there is a comment about context (e.g. why) the work was done if there isn't an issue to reference.
2. It may be good to have a reminder that if a gatekeeper has issues with a PR, they should add a comment and "request changes." This way the PR doesn't show up in the list for other gatekeepers.

PR #35452 (2023-04-18): Update pull request template
**Description of work.**

The pull request template has been updated to make it easier for developers and gatekeepers to review PRs. For full context and discussion on changes #35438 .

This PR makes the following changes to the template
- A section to record the purpose of the work undertaken. This is especially important for PRs with no associated issue
- A section to put a high level, brief (1-2 sentence) summary of the work done
- More instructions for giving a detailed description
- Testing instructions expanded for clarification
- Adds information for reviewe [... cut]
closed 2023-04-18 by peterfpeterson