mantidproject/mantid #35535: Manual Testing ISIS Indirect interfaces

SilkeSchomann, 2023-05-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

Follow the directions at: https://developer.mantidproject.org/Testing/IndirectInelastic/IndirectInelasticAcceptanceTests.html

gemmaguest, 2023-05-17:
Testing on IDAaaS

AnthonyLim23, 2023-05-18:
> * We got a "logic error in ConvertUnits" reported in item 13 of the first section under the Data Reduction section of the instructions and the plot did not appear.
> * We had a couple of attempts at the second section of the Data Reduction section but didn't get the expected workspaces. First time for items 11 and 16 each plot only had one spectrum rather than the number indicated. The second attempt the workspace indicated at line 10 didn't exist. This might be user error but for expediency we are skipping the rest of the section and it might be worth @AnthonyLim23 having a quick look to check it is ok.
> * In Data analysis MSD in step 13 only one of the fits succeeds.

Thank you @MohamedAlmaki for doing the first two.

For the last one, it seems ok for me. I expect some fits will fail (bad model for the data). I do think the doc needs updating .

MohamedAlmaki, 2023-05-22:
I opened issue #35607 for the documentation update. I don't think there is something else left.

sf1919, 2023-05-25:
Testing complete and new issues raised.

closed 2023-05-25 by sf1919