mantidproject/mantid #35536: Manual Testing Documentation

SilkeSchomann, 2023-05-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

Check Online Docs and the Qt-help docs built into MantidWorkbench (from the help droppdown menu)

* Algorithm, fit, concept and api pages should be generated
* Algorithm dialog snapshots should appear on algorithm pages in offline help
* Math formulae should appear on algorithm pages in offline help
* workflow diagrams should appear on algorithm pages in offline help

You may wish to use this script ( [OpenMostDocumentationForTesting.py](https://github.com/mantidproject/mantid/blob/master/tools/scripts/OpenMostDocumentationForTesting.py) ) to open all the online docs pages!

jhaigh0, 2023-05-17:
Didn't find any obvious Docs problems 👍 (Ubuntu)

jhaigh0, 2023-05-17:
In https://docs.mantidproject.org/nightly/interfaces/index.html#interfaces-contents

[image]

`feature_flags` should be in the F section

SilkeSchomann, 2023-05-17:
[image]
[image]

There seems to be a problem with rendering LaTeX on https://developer.mantidproject.org/Testing/DirectInelastic/MSliceTestGuide.html

closed 2023-05-18 by ConorMFinn