mantidproject/mantid #35537: Manual Testing Workbench Group 1- 4

SilkeSchomann, 2023-05-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/Core/Core.html

jhaigh0, 2023-05-17:
Report usage data warning box is confusing

jhaigh0, 2023-05-17:
Text collision on the concepts help page

[image]

SilkeSchomann, 2023-05-17:
Clicking `Mantid Help` in the `Help` menu causes a crash.

ajjackson, 2023-05-17:
A lot of the fonts available in the Figure Options dialogue (i.e. gear button from a plot) aren't available on IDAaaS. (Tested on a Molecular Spectroscopy instance.)

[image]

ajjackson, 2023-05-17:
Two issues with the normalisation feature of "Superplot"

- It is a little ambiguous from the name or the documentation that this will normalise every curve individually

> By using the dedicated checkbox above this list, one can toggle the activation of the normalisation (by the maximum value) of the curves.

could also be interpreted as meaning that all curves will be scaled by the same amount (i.e. from the maximum value across all curves)

- If the superplot button is clicked twice to close and re-open that section, the normalisation remains active but the box is no longer ticked

[image]

SilkeSchomann, 2023-05-17:
This link does not work: https://developer.mantidproject.org/Testing/Core/WorkbenchScriptRepository
(from https://developer.mantidproject.org/Testing/Core/Core.html)

ajjackson, 2023-05-17:
Running Abins with "SaveAscii" enabled creates a lot of hidden workspaces. (This seems to be a side-effect of calling SaveAscii but does not happen when SaveAscii is called directly. One for me to look into later...)

jhaigh0, 2023-05-17:
Link to script repository docs on the core testing page gives a 404 https://developer.mantidproject.org/Testing/Core/WorkbenchScriptRepository

ajjackson, 2023-05-17:
Created a notebook file with GenerateIPythonNotebook and opened with Jupyter Notebook in the same Conda environment. Immediately there was a warning about the file being updated to a new version - but that seems ok?

However after running the cells and waiting a minute, this message popped up:

[image]

ConorMFinn, 2023-05-17:
In a Show Data window, right clicking a column header to plot a bin or a row header to plot a spectrum does not plot the whole bin/spectra unless it is clicked to select it first. Which isn't very intuitive.

ConorMFinn, 2023-05-17:
Getting a slightly odd error from the script here: https://docs.mantidproject.org/nightly/concepts/MatrixWorkspace.html#workspace-algebra
```
Plus successful, Duration 0.00 seconds
Unlocking w3
Unlocking INTER00013462
Unlocking INTER00013463
CreateSingleValuedWorkspace started
Algorithm: CreateSingleValuedWorkspace v1
Execution Date: 2023-05-17 11:27:22.200648000
Execution Duration: -1 seconds
UUID: e0b3e2da-bdbf-4f7e-9c39-e4c6b90c2df3
Parameters:
  Name: OutputWorkspace, Value: __python_binary_op_single_value, Default?: No, Direction: Output
  Name: DataValue, Value: 2, Default?: No, Direction: Input
  Name: ErrorValue, Value: 0, Default?: Yes, Direction: Input
Time to validate properties: 0.000003 seconds
Time for other input validation: 0.000000 seconds
Time for other initialization: 0.000424 seconds
Time to run exec: 0.000178 seconds

CreateSingleValuedWorkspace successful, Duration 0.00 seconds
Add Data Object __python_binary_op_single_value successful
Multiply started
Algorithm: Multiply v1
Execution Date: 2023-05-17 11:27:22.201615000
Execution Duration: -1 seconds
UUID: 8358914a-92e3-484f-97ac-117acb963794
Parameters:
  Name: LHSWorkspace, Value: w3, Default?: No, Direction: Input
  Name: RHSWorkspace, Value: __python_binary_op_single_value, Default?: No, Direction: Input
  Name: OutputWorkspace, Value: w4, Default?: No, Direction: Output
  Name: AllowDifferentNumberSpectra, Value: 0, Default?: Yes, Direction: Input
  Name: ClearRHSWorkspace, Value: 0, Default?: Yes, Direction: Input
Write-locking w4
Read-locking w3
Read-locking __python_binary_op_single_value
Data Object 'w4' replaced in data service.

Time to validate properties: 0.000002 seconds
Time for other input validation: 0.000000 seconds
Time for other initialization: 0.000603 seconds
Time to run exec: 0.000343 seconds

Multiply successful, Duration 0.00 seconds
Unlocking w4
Unlocking w3
Unlocking __python_binary_op_single_value
Error occurred in handler:
Traceback (most recent call last):
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/observers/ads_observer.py", line 23, in wrapper
    func(*args, **kwargs)
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/observers/ads_observer.py", line 60, in deleteHandle
    self.presenter.close(name)
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/observers/observing_presenter.py", line 32, in close
    if self.current_workspace_equals(workspace_name):
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/observers/observing_presenter.py", line 50, in current_workspace_equals
    return self.model.workspace_equals(name)
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/workspacedisplay/table/model.py", line 151, in workspace_equals
    return self.ws.name() == workspace_name
RuntimeError: Variable invalidated, data has been deleted.
Data Object '__python_binary_op_single_value' deleted from data service.
```

ConorMFinn, 2023-05-17:
Algorithm Details window appearing behind mantid main window on macOS

ConorMFinn, 2023-05-18:
Issues Created:
- #35573
- #35574 
- #35575
- #35576
- #35578
- #35582
- #35583
- #35584

Existing Issues:
- https://github.com/mantidproject/mantid/issues/34739 (Tom looking into this one, but it's not a priority as we released with it last time)

closed 2023-05-18 by ConorMFinn