mantidproject/mantid #35538: Manual Testing Muon

SilkeSchomann, 2023-05-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

See instructions at: https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/index.html
https://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html

gemmaguest, 2023-05-17:
In MUSR instructions I could not find the `select plot data` button so could not test that part. I also didn't know what the `data loading pop-up` was under the grouping tab, or how to "change the value of Group 1". https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_Analysis_MUSR.html#muon-analysis-musr-ref

gemmaguest, 2023-05-17:
I can't test PSI because the data does not seem to be in the sample data or archive: https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_Analysis_PSI.html#muon-analysis-psi-ref

gemmaguest, 2023-05-17:
In HIFI I got an error changing the plot dropdown to counts, it said "Pair workspace have no counts workspace, plotting Asymmetry".
https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_HIFI.html#muon-analysis-hifi-ref

gemmaguest, 2023-05-17:
For ALC peak fitting I got this error. I don't think I understand where the sections shoudl be.
[image]

https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_ALC.html#muon-alc-ref

gemmaguest, 2023-05-17:
These instructions seem largely a duplicate of EMU and possibly others so I have not done them. It would be good to clarify which ones should be included: https://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html

gemmaguest, 2023-05-17:
Testing for everything else complete without problems.

AnthonyLim23, 2023-05-18:
I will have a look at these this afternoon. I think some of it is just doc updates

AnthonyLim23, 2023-05-18:
> I could not find the `se`

Unfortunately the only PSI data we have is from a user, so we cannot publicly share it.

AnthonyLim23, 2023-05-18:
> These instructions seem largely a duplicate of EMU and possibly others so I have not done them. It would be good to clarify which ones should be included: https://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html

I think this page has been deleted from the code base (I can't find it) as it is no longer needed.

AnthonyLim23, 2023-05-18:
Other than the one bug, everything else looks like docs being unclear. I have a PR to fix them here https://github.com/mantidproject/mantid/pull/35591

I also checked that the PSI and ARGUS stuff worked as expected.

AnthonyLim23, 2023-05-22:
I'm happy that everything that needs an issue has one. However, its best to check with the team that was assigned the testing

closed 2023-05-26 by gemmaguest