mantidproject/mantid #35543: Manual Testing Sample Transmission calculator

SilkeSchomann, 2023-05-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

Make sure that inputs and outputs work sensibly, stress test with some bad inputs (e.g. letters in a numeric input)
See https://docs.mantidproject.org/interfaces/general/Sample%20Transmission%20Calculator.html

rbauststfc, 2023-05-17:
Tested a little on both Windows and Linux, with some parallel testing on IDAaaS too. We found no problems had a few usability suggestions that could be picked up in future releases (and may be good induction issues):

- We felt the visibility of the some of the error messages could potentially be improved - it was easy to miss the red message at the bottom of the dialog.
- A tooltip to explain the format to enter into the `Multiple` input wavelength range type (which takes a rebin parameter string) would be helpful, as we didn't find this very intuitive.
- Including axis labels on the plot might be helpful.

Testing took approximately 20 minutes.

rbauststfc, 2023-05-18:
I've added a single issue to cover all of these (#35586) as it didn't seem necessary to create separate issues for each one, even if the work is ultimately split into separate PRs.

closed 2023-05-18 by rbauststfc