mantidproject/mantid #35544: Manual Testing ISIS Reflectometry

SilkeSchomann, 2023-05-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

http://developer.mantidproject.org/Testing/ReflectometryGUI/ReflectometryGUITests.html

gemmaguest, 2023-05-17:
One minor inconsistency on the Preview Tab behaviour: if you manually edit the ROI in the Experiment Settings tab, any processed runs get their state reset (i.e. reset from green to white). However, if you change the ROI via the Apply button on the Preview tab, the state is not reset.

Otherwise, all looks good :+1:

gemmaguest, 2023-05-17:
Raised issue #35565 for the one problem found so closing this.

closed 2023-05-17 by gemmaguest