mantidproject/mantid #35546: Manual Testing Live Data ISIS

SilkeSchomann, 2023-05-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/LiveData/LiveDataTests.html

MohamedAlmaki, 2023-05-17:
After clicking run in live histogram data, mantid crashes and gives the following error:
```
********* UNHANDLED EXCEPTION *********
  what(): Unknown property search object BufferThreshold
Backtrace:
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/mantid/api/../../../../libMantidAPI.so(+0x1f441a) [0x7f0e9c1b841a]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/mantid/api/../../../../libMantidAPI.so(+0x100037) [0x7f0e9c0c4037]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/mantid/kernel/../../../../libstdc++.so.6(+0xb0514) [0x7f0ea0af2514]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/mantid/kernel/../../../../libstdc++.so.6(+0xb0566) [0x7f0ea0af2566]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/mantid/kernel/../../../../libstdc++.so.6(__cxa_rethrow+0) [0x7f0ea0af2758]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/mantid/kernel/../../../../libMantidKernel.so(+0x133c2d) [0x7f0ea0ffbc2d]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/mantidqt/../../../libMantidQtWidgetsCommonQt5.so(_ZNK8MantidQt3API15AlgorithmDialog20getAlgorithmPropertyERK7QString+0xfc) [0x7f0e599eb94c]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/mantidqt/../../../libMantidQtWidgetsCommonQt5.so(_ZNK8MantidQt3API15AlgorithmDialog13getInputValueERK7QString+0xbc) [0x7f0e599ebb1c]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/mantidqt/../../../libMantidQtWidgetsCommonQt5.so(_ZN8MantidQt3API15AlgorithmDialog16setPropertyValueERK7QStringb+0x49) [0x7f0e599f1b69]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/mantidqt/../../../libMantidQtWidgetsCommonQt5.so(_ZN8MantidQt3API15AlgorithmDialog17setPropertyValuesERK11QStringList+0x1fe) [0x7f0e599f253e]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/mantidqt/../../../libMantidQtWidgetsCommonQt5.so(_ZN8MantidQt3API15AlgorithmDialog6acceptEv+0x40) [0x7f0e599f4980]
 /mnt/ceph/home/l1115324/.conda/envs/mantidnightly_6.7/plugins/qt5/libMantidQtWidgetsPluginsAlgorithmDialogsQt5.so(+0x2d89c) [0x7f0e50c0789c]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/PyQt5/../../../libQt5Core.so.5(+0x1dfff5) [0x7f0e5dcf1ff5]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(_ZN15QAbstractButton7clickedEb+0x32) [0x7f0e5a407162]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(+0x202b7a) [0x7f0e5a407b7a]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(+0x20372c) [0x7f0e5a40872c]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(_ZN15QAbstractButton17mouseReleaseEventEP11QMouseEvent+0x79) [0x7f0e5a4087a7]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(_ZN7QWidget5eventEP6QEvent+0x134) [0x7f0e5a399202]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(_ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent+0xd9) [0x7f0e5a36f087]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(_ZN12QApplication6notifyEP7QObjectP6QEvent+0x663) [0x7f0e5a37336d]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/PyQt5/QtWidgets.abi3.so(+0x351a9e) [0x7f0e5aa91a9e]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/PyQt5/../../../libQt5Core.so.5(_ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent+0x8e) [0x7f0e5dcd4a18]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(_ZN19QApplicationPrivate14sendMouseEventEP7QWidgetP11QMouseEventS1_S1_PS1_R8QPointerIS0_Ebb+0x283) [0x7f0e5a372981]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(+0x1a338d) [0x7f0e5a3a838d]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(+0x1a4505) [0x7f0e5a3a9505]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(_ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent+0xd9) [0x7f0e5a36f087]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/PyQt5/QtWidgets.abi3.so(+0x351a9e) [0x7f0e5aa91a9e]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/PyQt5/../../../libQt5Core.so.5(_ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent+0x8e) [0x7f0e5dcd4a18]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/PyQt5/../../../libQt5Gui.so.5(_ZN22QGuiApplicationPrivate17processMouseEventEPN29QWindowSystemInterfacePrivate10MouseEventE+0x6b6) [0x7f0e5afe556e]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/PyQt5/../../../libQt5Gui.so.5(_ZN22QWindowSystemInterface22sendWindowSystemEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE+0x57) [0x7f0e5afd2091]
 /mnt/ceph/home/l1115324/.conda/envs/mantidnightly_6.7/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x5672c) [0x7f0e4fdbc72c]
 /home/l1115324/.conda/envs/mantidnightly_6.7/lib/python3.8/site-packages/PyQt5/../../.././libglib-2.0.so.0(g_main_context_dispatch+0x2ad) [0x7f0e5b8de2dd]
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-l1115324'
errorreports-[Notice] No information shared
errorreports-[Error] Continue working.
```

MohamedAlmaki, 2023-05-17:
We couldn't do the remaining tests because the live data window always crashes mantid.

gemmaguest, 2023-05-17:
It is working for me in histo mode but I get a similar error but for a different property for Event mode. The `SpectraList` property only seems applicable for histo mode (from a brief look at the code) but that input appears when you toggle to Event mode and disappears if you toggle back to histo mode! I wonder if there is some problem setting which inputs are enabled for each mode.

```
StartLiveData-[Error] Error in execution of algorithm StartLiveData:
StartLiveData-[Error] Unable to connect listener [ISISHistoDataListener] to [127.0.0.1:56789]: Exception

********* UNHANDLED EXCEPTION *********
  what(): Unknown property search object SpectraList

Backtrace:
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/mantid/api/../../../../libMantidAPI.so(+0x1f441a) [0x7f35cd9f441a]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/mantid/api/../../../../libMantidAPI.so(+0x100037) [0x7f35cd900037]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/./libstdc++.so.6(+0xb0514) [0x7f35d48fc514]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/./libstdc++.so.6(+0xb0566) [0x7f35d48fc566]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/./libstdc++.so.6(__cxa_rethrow+0) [0x7f35d48fc758]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/mantid/kernel/../../../../libMantidKernel.so(+0x133c2d) [0x7f35d2533c2d]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqt/../../../libMantidQtWidgetsCommonQt5.so(_ZNK8MantidQt3API15AlgorithmDialog20getAlgorithmPropertyERK7QString+0xfc) [0x7f354a3b294c]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqt/../../../libMantidQtWidgetsCommonQt5.so(_ZNK8MantidQt3API15AlgorithmDialog13getInputValueERK7QString+0xbc) [0x7f354a3b2b1c]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqt/../../../libMantidQtWidgetsCommonQt5.so(_ZN8MantidQt3API15AlgorithmDialog16setPropertyValueERK7QStringb+0x49) [0x7f354a3b8b69]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqt/../../../libMantidQtWidgetsCommonQt5.so(_ZN8MantidQt3API15AlgorithmDialog17setPropertyValuesERK11QStringList+0x1fe) [0x7f354a3b953e]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqt/../../../libMantidQtWidgetsCommonQt5.so(_ZN8MantidQt3API15AlgorithmDialog6acceptEv+0x40) [0x7f354a3bb980]
 /home/gemma/Downloads/mantidworkbenchnightly/plugins/qt5/libMantidQtWidgetsPluginsAlgorithmDialogsQt5.so(+0x2d89c) [0x7f3549cdc89c]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/PyQt5/../../../libQt5Core.so.5(+0x1dfff5) [0x7f354e5dfff5]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(_ZN15QAbstractButton7clickedEb+0x32) [0x7f354ae02162]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(+0x202b7a) [0x7f354ae02b7a]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(+0x20372c) [0x7f354ae0372c]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(_ZN15QAbstractButton17mouseReleaseEventEP11QMouseEvent+0x79) [0x7f354ae037a7]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(_ZN7QWidget5eventEP6QEvent+0x134) [0x7f354ad94202]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(_ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent+0xd9) [0x7f354ad6a087]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(_ZN12QApplication6notifyEP7QObjectP6QEvent+0x663) [0x7f354ad6e36d]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/PyQt5/QtWidgets.abi3.so(+0x351a9e) [0x7f354b551a9e]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/PyQt5/../../../libQt5Core.so.5(_ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent+0x8e) [0x7f354e5c2a18]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(_ZN19QApplicationPrivate14sendMouseEventEP7QWidgetP11QMouseEventS1_S1_PS1_R8QPointerIS0_Ebb+0x283) [0x7f354ad6d981]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(+0x1a338d) [0x7f354ada338d]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(+0x1a4505) [0x7f354ada4505]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/PyQt5/../../../libQt5Widgets.so.5(_ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent+0xd9) [0x7f354ad6a087]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/PyQt5/QtWidgets.abi3.so(+0x351a9e) [0x7f354b551a9e]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/PyQt5/../../../libQt5Core.so.5(_ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent+0x8e) [0x7f354e5c2a18]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/PyQt5/../../../libQt5Gui.so.5(_ZN22QGuiApplicationPrivate17processMouseEventEPN29QWindowSystemInterfacePrivate10MouseEventE+0x6b6) [0x7f354b90956e]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/PyQt5/../../../libQt5Gui.so.5(_ZN22QWindowSystemInterface22sendWindowSystemEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE+0x57) [0x7f354b8f6091]
 /home/gemma/Downloads/mantidworkbenchnightly/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x5672c) [0x7f3540b2572c]
 /home/gemma/Downloads/mantidworkbenchnightly/lib/python3.8/site-packages/PyQt5/../../.././libglib-2.0.so.0(g_main_context_dispatch+0x2ad) [0x7f359a88b2dd]
```

gemmaguest, 2023-05-17:
Raised #35567 for the crashing problem. I'll close this for now to indicate we've done all we can for now but will note on the issue that this needs re-testing once resolved.

closed 2023-05-17 by gemmaguest