mantidproject/mantid #35550: Manual Testing Error Reporter

SilkeSchomann, 2023-05-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

http://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ErrorReporterTesting.html

SilkeSchomann, 2023-05-17:
The documentation for the error reporter manual tests needs updating as it still mentions Daniel Murphy.

SilkeSchomann, 2023-05-17:
When running SegFault there is no `Show More Details` button but a `Python has stopped working` dialogue.

ConorMFinn, 2023-05-17:
All working as expected on macOS

ConorMFinn, 2023-05-18:
Issues found unable to be reproduced.