mantidproject/mantid #35551: Manual Testing Sliceviewer

SilkeSchomann, 2023-05-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html

robertapplin, 2023-05-17:
A couple of things to note from testing so far:
- It would be nice to have a tight layout option on the SliceViewer so that we can ask it not to get distorted
- It is not possible to export a EventWorkspace2D cut  - there is already a known issue open for this
- Exporting cuts on MDWorkspace uses the same number of bins on the whole slice rather than just a fraction of the number of bins

robertapplin, 2023-05-17:
This deprecation warning on step 3 of the 3rd test in the `Specific Tests` section (Multiplying the workspace from the IPython console)

```
Passing the angle parameter of __init__() positionally is deprecated since Matplotlib 3.6;
the parameter will become keyword-only two minor releases later
```

rbauststfc, 2023-05-17:
Tested on IDAaaS. Testing took approximately 1 hour and 20 mins.

RichardWaiteSTFC, 2023-05-18:
> A couple of things to note from testing so far:
> 
> * It is not possible to export a EventWorkspace2D cut  - there is already a known issue open for this

This is issue #33394

RichardWaiteSTFC, 2023-05-18:
Have created the following issues for milestone 6.8:
#35571 (cursor out of range error)
#35580 (matplotlib deprecation)
#35572 (a maintenance issue for updating mistakes in the testing docs)

There are a couple of nice to haves that I haven't created issues for because they are low priorities:
- Tight layout option on the SliceViewer so as not to crop axes labels
- Exporting cuts on MDWorkspace uses the same number of bins on the whole slice rather than just a fraction of the number of bins

@rbauststfc is that alright? Are you happy to close the issue?

rbauststfc, 2023-05-18:
Looks great, thanks @RichardWaiteSTFC! I'll close the issue now.

closed 2023-05-18 by rbauststfc