mantidproject/mantid #35553: Manual Testing ALF

SilkeSchomann, 2023-05-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/DirectInelastic/ALFViewTests.html

jhaigh0, 2023-05-17:
In 'Calculating the rotation angle' I got a rotation of 1.26828 instead of 0.675824 listed in the test.

SilkeSchomann, 2023-05-17:
It is also 1.26828 on Windows.

ConorMFinn, 2023-05-17:
@robertapplin Could you take a look at this one when you have a chance to see if this is a sensible value difference?

robertapplin, 2023-05-17:
Thanks for spotting this. The correct value is `1.26828`. It looks like I forgot to update the testing instructions after fixing this for the 6.6 release https://github.com/mantidproject/mantid/pull/35169

I have opened a PR to update this value in the documentation https://github.com/mantidproject/mantid/pull/35564

PR #35564 (2023-05-30): Update ALFView test docs
**Description of work**
This PR updates the ALFView manual testing documentation to provide the correct expected value for the rotation angle. I forgot to update this value when I made the fix in https://github.com/mantidproject/mantid/pull/35169 (before the release of Mantid 6.6).

**To test:**
Read the docs to make sure it makes sense

Fixes #35553

---

#### Reviewer ####

Please comment on the points listed below ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)).
**Your comments will be used as part of the gatekeeper process, so pl [... cut]
closed 2023-05-17 by jhaigh0