mantidproject/mantid #35567: Live data crashing

gemmaguest, 2023-05-17:
**Describe the bug**
Live data seems to be crashing due to invalid input properties. We should investigate if this is a regression but this should work so it is likely an urgent fix. The problem was found during manual testing using the TEST_LIVE facility but I have also seen it with a real instrument.

**To Reproduce**
See the instructions and comments on #35546

**Expected behavior**
It should not be possible to set invalid properties that cause Mantid to crash when running live data

**Platform/Version (please complete the following information):**
 - OS: Seen on Ubuntu and Windows, probably all
 - 6.6.20230516.1658

**Additional context**
First, it would be good to check if this is reproducible in 6.6 and/or establish when the problem was introduced before we decide to go ahead with a fix for 6.7. It looks serious though so we should investigate this as a priority.

On Windows, Mohammed and Mial could not get any of the test cases from the unscripted testing to work.

On Ubuntu 22.04, it seemed to work ok for me for histo but crashed after toggling to Event mode and trying to run that. As per the comments on the test issue, it seems to be getting invalid input properties somehow (but different ones in each case documented).

MohamedAlmaki, 2023-05-19:
I ran another test on Mantid 6.6 for Windows, comparing it to the current release. The tests were fine in the previous release but in the current release mantid crashes. It appears to be a regression issue.

jhaigh0, 2023-05-19:
What I am getting is an unknown property exception coming from where the `StartLiveDataDialog` algorithm runs `AlgorithmDialog::accept()`. This then calls `setPropertyValue` on each property from `m_algProperties` in `AlgorithmDialog::setPropertyValues`.

Eventually you end up in `PropertyManager::getPointerToProperty` which is passed the property name `"BufferThreshold"` which isn't in `m_properties` so an exception is thrown.

The only place this property is declared is in the constructor of `KafkaEventListener` which gets called once when you open the live data dialogue and once when you set the instrument property (I think).

So at least for me that's the skeleton of what's causing the crash, and I have a huge sense of déjà vu that I've looked at this exact problem before. Possibly it was something Daniel was telling me about before he left, I'll see if I can find anything.

mantid-builder, 2023-05-30:
Fixed by #35622

closed 2023-05-30 by mantid-builder