mantidproject/mantid #35572: Update manual testing docs for sliceviewer

RichardWaiteSTFC, 2023-05-18:
Found during testing of #35551 

- For 4D MD workspace, if you have a non-momentum viewing axis, the peak overlay won't update the slice point
- The peak actions don't work with a non-momentum viewing axis
- For the 3D MDHistoWorkspace in the instructions, it is not possible to open the SliceViewer
- For the CutViewer instructions, the non-orthogonal view should not be disabled
- For step 19, you need to move the cut representation

sf1919, 2023-06-23:
The screen shot for CutViewer Tool step 8 may be wrong. I get the following
[image]

sf1919, 2023-06-23:
Trying to follow the ADS observer for peak overlay and do not appear to be able to rename the file in the ADS

PR #36034 (2023-09-27): Update sliceviewer testing instructions
**Description of work**
Update sliceviewer testing instructions from feedback in #35551
Addressed all the points in the original issue except

- For the 3D MDHistoWorkspace in the instructions, it is not possible to open the SliceViewer

which I could not reproduce

**To test:**

(1) Check doc tests run
(2) Test during manual testing for 6.8 release

Fixes #35572 

---

#### Reviewer ####

Please comment on the points listed below ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)).
**Your comments will be used as part of  [... cut]
closed 2023-09-27 by SilkeSchomann