mantidproject/mantid #35574: Text Collides on Concepts Web Docs Page

ConorMFinn, 2023-05-18:
**Describe the bug**
Colliding of text on the web version of the Concepts index page (The qt one in workbench is fine). 
[image]

**To Reproduce**
1. Visit: https://docs.mantidproject.org/nightly/concepts/index.html#concepts-contents

**Expected behavior**
The link should pass onto the next line with a hyphen, as in the qt docs within workbench. 

**Platform/Version (please complete the following information):**
 - OS: All

sf1919, 2025-03-12:
Moving this to v6.14 as other work on help documentation being undertaken in 6.13

sf1919, 2026-01-15:
@peterfpeterson as you were involved in the original page can you look into this? Might be a quick win?