mantidproject/mantid #35575: Fonts Available in Plot Options not Available on System

ConorMFinn, 2023-05-18:
**Describe the bug**
A lot of the fonts available in the Figure Options dialogue (i.e. gear button from a plot) aren't available on IDAaaS. (Tested on a Molecular Spectroscopy instance.)

**To Reproduce**
1. Boot up an IDAaaS instance
2. Plot something
3. Open the figure options (cog button) on the plot
4. Switch to the legend tab

**Expected behavior**
1. Only available plots should be available in the font list.

**Screenshots**
[image]

**Platform/Version (please complete the following information):**
 - OS: CentOS 7
 - OS Version: IDAaaS
 - Mantid Version 6.7 Nightly

**Additional context**

jclarkeSTFC, 2023-10-09:
I've just tried this on IDAaaS workspaces for a couple of different instruments with Mantid 6.7 and I couldn't reproduce it. However I also found [this](https://stackoverflow.com/questions/26085867/matplotlib-font-not-found), which might be useful if anyone tries to fix this.

ConorMFinn, 2023-10-13:
This appears to have fixed itself, may have just been found at a point between mpl/os versions where the default setting for the font was one that wasn't available. Regardless, I'm happy to close this until it crops up in testing again

closed 2023-10-13 by ConorMFinn