mantidproject/mantid #35576: Superplot: Normalise by Max Issues

ConorMFinn, 2023-05-18:
**Describe the bug**
Two issues with the "Normalise by max" feature of "Superplot":

- [ ] It is a little ambiguous from the name or the documentation that this will normalise every curve individually

  From the docs:
  > By using the dedicated checkbox above this list, one can toggle the activation of the normalisation (by the maximum value) of the curves.

  This could also be interpreted as meaning that all curves will be scaled by the same amount (i.e. from the maximum value across all curves)

- [ ] If the superplot button is clicked twice to close and re-open that section, the normalisation remains active but the box is no longer ticked

**To Reproduce**
1. Plot multiple spectra 
2. Click the superplot button
3. Enable normalisation
4. Click the superplot button to close it
5. Reopen the superplot menu

**Expected behavior**
- The documentation or tooltips should be made more clear
- The state of the checkbox should be maintained between closing and opening the superplot

**Screenshots**
[image]

**Platform/Version (please complete the following information):**
 - OS: All

thomashampson, 2023-09-14:
I've added some issues from 6.8 manual testing

sf1919, 2024-01-22:
Has this been completed or does the milestone need changing?