mantidproject/mantid #35580: Matplotlib deprecation warning in sliceviewer

RichardWaiteSTFC, 2023-05-18:
**Describe the bug**
Found during manual testing #35551

**To Reproduce**
This deprecation warning on step 3 of the 3rd test in the Specific Tests section of manual testing docs  (Multiplying the workspace from the IPython console)
https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#ads-observer-for-workspace

```

Passing the angle parameter of __init__() positionally is deprecated since Matplotlib 3.6;
the parameter will become keyword-only two minor releases later

```

jhaigh0, 2023-07-20:
@RichardWaiteSTFC Just to check, is it the line `mtd['md_3Dim'] *= 2` that was causing the deprecation warning? It doesn't seem to do for me.

RichardWaiteSTFC, 2023-07-20:
Yes that's the line that triggered the warning (with sliceviewer still open) - it might have been in non-orthogonal view (just thinking about the angle parameter in question)?

jhaigh0, 2023-07-20:
Ah okay, I'll try following the instructions a little ore exactly this time.

thomashampson, 2023-08-11:
@RichardWaiteSTFC I've tried following the test instructions and fiddling with other things and I can't reproduce the depreciation warning.

PR #36511 (2023-12-05): Fix mpl deprecation warning from the slice viewer
### Description of work

The `angle` argument for the `matplotlib.patches.Ellipse` class will become keyword only in version 3.8
https://github.com/matplotlib/matplotlib/blob/v3.7.3/lib/matplotlib/patches.py#L1522

Added the keyword in to stop it warning us.

I also checked for any other positional patch arguments in the codebase which are deprecated, but this seems to be the only one.

Fixes #35580 

### To test:

The issue was spotted during manual testing (see issue #35580) but I could never replicate it (I just looked for similar cases onli [... cut]
closed 2023-12-05 by SilkeSchomann