mantidproject/mantid #35584: Workspace Algebra Hidden Workspace Deleted

ConorMFinn, 2023-05-18:
**Describe the bug**
Running a script from the algebra docs causes this error to appear:

```
Unlocking __python_binary_op_single_value
Error occurred in handler:
Traceback (most recent call last):
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/observers/ads_observer.py", line 23, in wrapper
    func(*args, **kwargs)
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/observers/ads_observer.py", line 60, in deleteHandle
    self.presenter.close(name)
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/observers/observing_presenter.py", line 32, in close
    if self.current_workspace_equals(workspace_name):
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/observers/observing_presenter.py", line 50, in current_workspace_equals
    return self.model.workspace_equals(name)
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/workspacedisplay/table/model.py", line 151, in workspace_equals
    return self.ws.name() == workspace_name
RuntimeError: Variable invalidated, data has been deleted.
```

**To Reproduce**
Run this script:
```
w1 = mtd['workspace1']
w2 = mtd['workspace2']

# Sum the two workspaces and place the output into a third
w3 = w1 + w2

# Multiply the new workspace by 2 and place the output into a new workspace
w4 = w3 * 2
```

**Expected behavior**
No error should appear.

jhaigh0, 2023-05-19:
When I run the script, is says `workspace1` doesn't exist, which I think makes sense?
I loaded two workspaces and tried the rest of the script with them, and it worked with no errors. Maybe I'm misunderstanding the issue.

sf1919, 2024-01-22:
@cailafinn and @jhaigh0 is this still an issue?