mantidproject/mantid #35592: The indirect testing docs need updating

AnthonyLim23, 2023-05-18:
The testing docs do not take into account the new inelastic interface. It might be worth making a seperate set of testing docs that use both direct and indirect data. https://developer.mantidproject.org/Testing/IndirectInelastic/IndirectInelasticAcceptanceTests.html

What do you think @SilkeSchomann ?

jhaigh0, 2023-09-11:
this may be solved by #36032

MohamedAlmaki, 2023-09-11:
@SilkeSchomann Could you provide feedback on this? 
There are currently two tests: one for Elwin and the other for IQT. I could copy the current docs into another file and then add the same tests but using direct data

SilkeSchomann, 2023-09-11:
@MohamedAlmaki I would move the tests for Elwin and IQT to a new Inelastic test document and provide data for both direct and indirect. It will probably be easier to spot bugs this way.

MohamedAlmaki, 2023-09-11:
The problem is that there are some tests that depend on the output of these tests

SilkeSchomann, 2023-09-12:
Ok, in that case let's implement your initial suggestion and just copy them to be used with direct data.

MohamedAlmaki, 2023-09-14:
What is the direct run equivalent of irs26176_graphite002_red.nxs? Also, can I get a resolution file like irs26176_graphite002_red.nxs?

AnthonyLim23, 2023-09-14:
The workspace name is <instrument><run number>_<analyser><reflection of analyser>_<data type>

where the data type is normally red = reduced or res = resolution.

MohamedAlmaki, 2023-09-14:
Sorry, what is the name?

AnthonyLim23, 2023-09-14:
sorry, it didnt show up because I forgot the code mark things

PR #36479 (2023-12-11): Add direct inelastic tests for data manipulation tab
### Description of work
This pull request introduces new manual testing documentation for the inelastic data manipulation tab, which was previously only tested with indirect data. It uses the same instruction used for indirect data.

The PR also does a small bit of re-organised the manual testing directories for direct/indirect/inelastic so that it matches the interface category directory structure in Mantid. The links used in the manual testing issues will be updated in this PR: https://github.com/mantidproject/documents/pull/118

Fixes #35 [... cut]