mantidproject/mantid #35597: Error in GSAS2 tab of EngDiff UI when adjusting fitting range

RichardWaiteSTFC, 2023-05-19:
**Describe the bug**
Found during manual testing #35541 (not a regression)
Error in GSAS2 tab of EngDiff UI when adjusting fitting range

```
Traceback (most recent call last):  File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqtinterfaces/Engineering/gui/engineering_diffraction/tabs/gsas2/presenter.py", line 36, in on_refine_clicked 
number_output_histograms = self.model.run_model(  File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqtinterfaces/Engineering/gui/engineering_diffraction/tabs/gsas2/model.py", line 115, in run_model 
self.load_basic_outputs(gsas_result)  File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqtinterfaces/Engineering/gui/engineering_diffraction/tabs/gsas2/model.py", line 692, in load_basic_outputs
self.create_lattice_parameter_table()  File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqtinterfaces/Engineering/gui/engineering_diffraction/tabs/gsas2/model.py", line 810, in create_lattice_parameter_table
with open(parameters_txt, "rt", encoding="utf-8") as file:FileNotFoundError: [Errno 2] No such file or directory: '/home/ra1112751/Engineering_Mantid/GSAS2/hello/hello_cell_parameters_Fe_gamma.txt'
```

**To Reproduce**
Test 11 of manual testing instructions
(https://developer.mantidproject.org/Testing/EngineeringDiffraction/EngineeringDiffractionTestGuide.html)

**Expected behavior**
Would pass fitting range to GSAS2 refine command and would retrieve results without error

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

RichardWaiteSTFC, 2023-05-25:
Also need to fix 2 related bugs:

1. Min limit > Max limit from fit range sliders or edit boxes
2. Generate reflections withing d-spacing range corresponding to TOF fit limits (or at least only pass those within limits as input to GSAS)

mantid-builder, 2023-06-05:
Fixed by #35619

PR #35619 (2023-06-05): Fix bugs related to fit limits in GSAS2 tab in EngDiff UI
**Description of work**
Fixes 2 bugs (not regressions - hence the release notes -  but it would be nice to get into 6.7 as they are easy to trigger):

1. Checks reflections file exists before trying to read it (this happens if no reflections in TOF limits - but there may be other reasons - e.g. if refinement fails?)
2. Stop users entering min > max using sliders or edit boxes (atm it doesn't do anything fancy it just resets the slider and box to the data min/max)

Long term it would be good to only generate reflections within limits of  [... cut]
closed 2023-06-05 by mantid-builder