mantidproject/mantid #35610: Muon crash with user function

AnthonyLim23, 2023-05-23:
from James

Opened Muon Analysis,
loaded a run (HIFI 180501) 
went to Fitting, 
added a Flat background, 
then tried to add a User Function. Crash!

AnthonyLim23, 2023-05-23:
```
Traceback (most recent call last):
  File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqtinterfaces/Muon/GUI/Common/fitting_widgets/tf_asymmetry_fitting/tf_asymmetry_fitting_presenter.py", line 54, in handle_function_structure_changed
    super().handle_function_structure_changed()
  File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqtinterfaces/Muon/GUI/Common/fitting_widgets/basic_fitting/basic_fitting_presenter.py", line 296, in handle_function_structure_changed
    self.view.set_current_dataset_index(self.model.current_dataset_index)
  File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqtinterfaces/Muon/GUI/Common/fitting_widgets/basic_fitting/basic_fitting_view.py", line 143, in set_current_dataset_index
    self.fit_function_options.set_current_dataset_index(dataset_index)
  File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqtinterfaces/Muon/GUI/Common/fitting_widgets/basic_fitting/fit_function_options_view.py", line 183, in set_current_dataset_index
    self.function_browser.setCurrentDataset(dataset_index)
Exception: unknown
```

AnthonyLim23, 2023-05-23:
the bug is in the previous release, if the user ignores the error reporter and carries on working it seems to be ok

AnthonyLim23, 2023-05-25:
This crash seems to be in all fit browser's except for the one in the plotting window

AnthonyLim23, 2023-05-26:
I have tracked it down to this line https://github.com/mantidproject/mantid/blob/main/qt/widgets/common/src/FunctionTreeView.cpp#L642

AnthonyLim23, 2023-05-26:
looks like the error comes from here https://github.com/mantidproject/mantid/blob/main/Framework/API/src/IFunction.cpp#L1403

the `m_attrs` is empty so `iter` is also empty and we cannot access `first`. It should have a value of `formula` for first.

mantid-builder, 2023-05-30:
Fixed by #35620

closed 2023-05-30 by mantid-builder