mantidproject/mantid #35665: Move Data analysis GUI

AnthonyLim23, 2023-06-08:
In the indirect GUI's there is a "data analysis" GUI. However, it is also used for direct experiments. This issue is to move it to the inelastic section (including files and docs updated).

MohamedAlmaki, 2023-10-19:
@SpencerHowells this is the issue for renaming. You said back in the email that the newly renamed subcategory 'Curve Fitting' should be present in both 'indirect' and 'inelastic' categories right?  
Because in this issue it is written to move the subcategory to inelastic not copying it from my understanding.

MohamedAlmaki, 2023-10-19:
Okay, right. Spencer suggested in an email today that the 'Curve Fitting' should be in both 'indirect' and 'inelastic' categories because it could be used for both direct and indirect data

AnthonyLim23, 2023-10-19:
inelastic is supposed to represent both, instead of the separate indirect and direct categories

MohamedAlmaki, 2023-10-19:
Ok, that makes sense

robertapplin, 2023-10-24:
We can do this in several steps, each in a different PR:

- [ ] 1. Move the Data Analysis interface to the `Inelastic` category. As part of this we want to remove it from the `Direct` and `Indirect` categories. We also want to make sure the documentation file for the interface is in the correct location.
- [ ] 2. Move the Data Analysis files to a new folder under `qt/scientific_interfaces/Inelastic/`. This includes moving the tests for this interface and creating a new target for Inelastic interfaces.
- [ ] 3. Rename the interface to "Curve Fitting" as described by Spencer. Rename the tabs. Rename the files and classes so that they match the new naming

robertapplin, 2023-11-03:
I have opened a separate issue for renaming the Data Analysis GUI so that this issue can be closed when step 2 is finished https://github.com/mantidproject/mantid/issues/36363

PR #36364 (2023-11-07): Moving Data Analysis GUI interface files to inelastic folder.
### Description of work
A new qt library / test library for the inelastic interface category has been created.
 A new inelastic subdirectory has been created,  and the data analysis interface files and test files have been moved 
to the new category.
#### Summary of work

Fixes #35665 
Fixes #36323

#### Further detail of work

### To test: