mantidproject/mantid #35677: Always output a decimal place for SpinEchoLength in SaveSESANS file

rbauststfc, 2023-06-14:
Requested by Greg Smith at ISIS.

The `SaveSESANS` function writes a `*.ses` file from normalised polarisation data from SESANS experiments. The first data column saved out is the `SpinEchoLength` column. If the number in this column rounds to a 0 after the decimal point, Mantid does not write out any decimal point.

When importing this data into one of the possible SESANS analysis packages as ASCII text (not as a *.ses file), lines without a training decimal are ignored. This means that there are gaps in the imported data. This is not the case when importing a *.ses file, but incorporating 3- or 4-column ASCII is one way that at least Greg imports the data.

Greg would like data in these files to output a .0 when the value rounds to 0 (i.e. a value of 13827 would be written out as 13827.0). I'm still confirming with Greg that this is only required for the `SpinEchoLength` column in the file, and that the number of decimal places for entries that don't round to 0 should remain unchanged.

Greg has asked the developer if the data importing routine can be amended to resolve this, but felt it would be good to address this from both angles if possible. It's considered to be a bit of an edge case, but nevertheless worthwhile for SESANS users.

smk78, 2023-06-23:
>Comment from Steve - we should check all writers as part of this. I believe this would mean `SaveSESANS`, `SaveCanSAS1D`, `SaveNXcanSAS` and `SaveRKH`.

Certainly in the XML files written by SaveCanSAS1D I have seen transmission values written without any decimals; ie, as:
```
<T unit="none">1</T>
```
This is easy to replicate: just make the Sample/Can Trans run # and the Sample/Can Direct run # the same (so you are dividing like by like and will get unity).

smk78, 2023-06-23:
Should probably also check the CSV writer. Users sometimes write this through the Save Other button.