mantidproject/mantid #35681: Workbench editor font appear with strange spacing

zjmorgan, 2023-06-14:
In Mantid Workbench, the fonts in the editor appear with strange spacing. 

[image]

This is being observed on RHEL7 but unaware of other affected platforms.

A temporary fix is to switch the default font in Mantid Workbench to a monospaced font then restart.

Last PR mentioning fonts.
https://github.com/mantidproject/mantid/pull/33138

**Describe the bug**

Odd, non-monospaced fonts appear in the editor

**To Reproduce**

Open Mantid Workbench, and open a new script with text in it. Look at the text for odd font spacings.

**Expected behavior**

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: RHEL 7
 - Mantid Version 6.7.0 release candidate

jhaigh0, 2023-06-15:
All seems okay on windows
[image]

peterfpeterson, 2023-06-15:
Another piece of the puzzle. On the same analysis computer (node12)
* connecting via `ssh -Y` brings up the workbench without the font issue.
* connecting via thinlinc exhibits the issue

The version of virtualgl installed is 3.0.2

peterfpeterson, 2023-06-15:
The issue can be solved by setting a default monospace font in [.../workbench/config/fonts.py](https://github.com/mantidproject/mantid/blob/main/qt/applications/workbench/workbench/config/fonts.py#L48-L58)

mantid-builder, 2023-06-16:
Fixed by #35689

PR #35689 (2023-06-16): Set font on non-ubuntu linux
On linuxes other than ubuntu, the fall through behavior for default font produced a non-monospace font on RHEL7. This fixes that by setting font options that are available on RHEL.

Additionally, the os inspection behavior was a repeat of functionality in `mantid.kernel.environment` and was changed to use that central version.

**To test:**

On the ornl analysis cluster, connect via thinlinc and bring up mantidworkbench. Before this patch, any file will have very strange spacing, after this one, everything should be spaced reasonably.

Fixes #35681. 

---

#### Rev [... cut]
closed 2023-06-16 by mantid-builder