mantidproject/mantid #35694: Create flexible build and test pipeline to shorten feedback loop

robertapplin, 2023-06-19:
**Description**
Recent efforts to debug issues on our nightly pipelines has shown we need to shorten the feedback loop if we want to make rapid progress. Our current pipelines can take hours before we know if a change we've made has fixed an issue. It would be very useful to have a pipeline where we can specify which part of a build step we want to run.

As an example, here are certain situation we might want to run:
- Only run system tests on OSX
- Only run unit tests on [select operating system]
- Only run doc tests on [select operating system]

As an example, this PR was created specifically for debugging purposes to reduce our feedback loop from 3-4 hours to 15 minutes:
https://github.com/mantidproject/mantid/pull/35683

**Describe alternatives you've considered**
A new pipeline could be created to provide greater flexibility for the build and test steps we run.

robertapplin, 2023-09-15:
Ah found it, I had a feeling I had created something similar to https://github.com/mantidproject/mantid/issues/35921 !

sf1919, 2024-07-05:
Docs tests were moved to linux as part of #37058

sf1919, 2024-07-05:
Closing this Issue as it is going to be resolved by #35921

closed 2024-07-05 by sf1919