mantidproject/mantid #35719: Mistake in the EISF for confined diffusion in cylinder geometry

fucaline, 2023-06-22:
**Describe the bug**
The implemented EISF for the diffusion of a particle confined in a cylinder:  Framework/PythonInterface/plugins/functions/EISFDiffCylinder.py does not follow the theoretical expression given in ref1: Dianoux et al. [Mol. Phys. 46:1129-37, 1982](https://doi.org/10.1080/00268978200101121).

- In EISFDiffCylinder.py: A0=[j0(QLcos(theta))]^2
- In ref1: A0=[j0(QLcos(theta)/2)]^2

- In EISFDiffCylinder.py: B0=[3*j1(QRsin(theta))/(QRsin(theta))]^2
- In ref1: B0=[2*J1(QRsin(theta))/(QRsin(theta))]^2

where J1 is an ordinary Bessel function of first kind, and j1 a spherical Bessel function.
Do j1 and J1 have been confused? Or 2*J1(x) have been approximated by 3*j1(x)?

Also, the description in the documentation page is incorrect: https://docs.mantidproject.org/nightly/fitting/fitfunctions/EISFDiffCylinder.html

- L and R have been exchanged
- A0 does not correspond to what is implemented in EISFDiffCylinder.py

robertapplin, 2023-12-21:
Keep open

robertapplin, 2024-01-03:
@fucaline Thanks for reporting this, and sorry we've taken so long to get back to you! I am investigating this now

So far I have read through the paper and found the equation they are using for the A0 parameter. I get it to be the same as that implemented in EISFDiffCylinder.py   i.e.  A0=[j0(QLcos(theta))]^2 . These are my workings, can you spot if I have made a mistake?

```
Assuming z = LQcos(\theta), and using equation 2b in the paper with p=0 I get

A_0(z) = \frac{2(1 - \cos(z))}{z^2}

I then use the cosine of half-angle identity \cos\left(\frac{x}{2}\right) = \sqrt{\frac{1 + \cos(x)}{2}} to get:

1 - \cos(z) = \frac{1}{2} \cdot (z \cdot j_0(z))^2

I substitute this into the A_0(z) equation, and I get:

A_0(z) = j_0(z)^2
```

robertapplin, 2024-01-04:
These are my workings for B0. From the paper we have:
[image]

We can then use the following formula
[image]

to find J_1 in terms of j:
[image]

Substitute this into the B0 equation and we get:
[image]

The j_{3/2}(z) can be calculated using `scipy.special.spherical_jn`

PR #36607 (2024-01-05): Update EISFDiffCylinder to use `scipy.special` Bessel functions
### Description of work
This PR fixes a mistake in the EISFDiffCylinder function for the `B_0` parameter. It looks like J_1 (the ordinary Bessel function of the first kind) was confused with j_1 (the spherical Bessel function). (although it is still unclear if this was intentional.)

There were a few problems outlined in the attached issue.
- It was suspected that the `A_0` parameter did not match the parameter provided in the paper. My workings [here](https://github.com/mantidproject/mantid/issues/35719#issuecomment-1875685703) s [... cut]
closed 2024-01-05 by SilkeSchomann