mantidproject/mantid #35730: Windows Workspace/Algorithm Smoke Tests   :man_astronaut: :woman_astronaut: :space_invader:

SilkeSchomann, 2023-06-23:
We must test both the .exe installer package and install direct from conda.

* To install via conda:
  - Use Intel Conda and make sure conda-forge is added to channels
```
mamba create -n test_mantid_6.6
mamba activate test_mantid_6.6
mamba install -c mantid/label/rc1 mantidworkbench --yes
workbench # to launch
```

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :man_astronaut: :woman_astronaut: :space_invader:

* Load some workspaces, try 5 from different instruments
* Show detectors on these workspaces
* Check Instrument Viewer for these workspaces, follow  documentation for  [Instrument Viewer Widget](https://docs.mantidproject.org/workbench/instrumentviewer.html) and some more detail on the picking tools in the [Basic Course](https://docs.mantidproject.org/tutorials/mantid_basic_course/connecting_data_to_instruments/03_investigating_data.html)
  - [ ] Render tab 
  - [ ] Draw tab 
  - [ ] Pick tab 
  - [ ] Instrument tree tab 
* Test 5 - 10 random algorithms, follow [how to run Algorithms](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/algorithms_workspaces_and_histories/02_algorithms.html#algorithms) and [pick your favourite](https://docs.mantidproject.org/algorithms/)
 - [ ] Help button 
 - [ ] Validation of inputs 
 - [ ] Overwriting workspace 
 - [ ] Run the algorithm 
- [ ] Check right-click options of the workspaces, they are listed [in the image in the documentation](https://docs.mantidproject.org/nightly/workbench/workspacetreewidget.html#workbenchworkspacetoolbox).

jhaigh0, 2023-06-23:
In the instrument view render tab, if I turn on the wireframe option (under display settings) in full 3D mode, the model only appears when clicking and dragging. That's using `MAR11060` from the training course data

jhaigh0, 2023-06-23:
(Can't test the help buttons because on Windows - Conda)

closed 2023-06-23 by SilkeSchomann