mantidproject/mantid #35739: Linux Clean/Dirty Install Smoke Tests  :soap: :hankey:

SilkeSchomann, 2023-06-23:
* To install the tar.xz package for Linux, run `(sudo) tar -xJf mantid-VA.B.C.tar.xz` in a terminal and it will unzip the package in your current working directory. 
* To install via conda:
  - Use Intel Conda and make sure conda-forge is added to channels
```
mamba create -n test_mantid_6.6
mamba activate test_mantid_6.6
mamba install -c mantid/label/rc1 mantidworkbench --yes
workbench # to launch
```

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :soap: :hankey:

### Dirty install

* Make sure that you have several versions of Mantid installed
 * Last release
 * A nightly
 * If possible an old release
* Install the new release candidate of Mantid
- [ ] Check that Mantid boots up correctly

### Clean install

* Remove all existing Mantid versions and associated files

**On Windows**:
 * Uninstall the program
 * Clear shortcuts from desktop
 * Remove the mantid  files in %APPDATA%

**On macOS** :
 * Remove the application
 * Remove the `~/.mantid directory`
 * Remove (or rename) `~/Library/Preferences/org.python.python.Python.plist`  and `~/Library/Preferences/org.python.python.plist` 

**On Linux** :
 * Remove the package: `/opt/Mantid`
 * Remove `~/.config/Mantid`
 * Remove `~/.mantid/`

Re-install the new release candidate of Mantid

- [ ] Check that Mantid boots up correctly

sf1919, 2023-06-23:
For future testing this smoke test is no longer relevant and probably should be removed

gemmaguest, 2023-06-23:
Install via conda worked fine on Ubuntu 22.04. Previous settings were picked up from `~/.mantid`; deleting that file meant a clean start up.

closed 2023-06-23 by gemmaguest