mantidproject/mantid #35741: Linux Interface Smoke Tests  :facepalm:

SilkeSchomann, 2023-06-23:
* To install the tar.xz package for Linux, run `(sudo) tar -xJf mantid-VA.B.C.tar.xz` in a terminal and it will unzip the package in your current working directory. 
* To install via conda:
  - Use Intel Conda and make sure conda-forge is added to channels
```
mamba create -n test_mantid_6.6
mamba activate test_mantid_6.6
mamba install -c mantid/label/rc1 mantidworkbench --yes
workbench # to launch
```

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :facepalm:

## ISIS only, if possible, so catalogue access can be tested

- [ ] MantidWorkbench opens without errors or warnings 
- [ ] Every option in `Interface` opens a GUI 
- [ ] Load some test data 
- [ ] Access Catalogue through algorithms (use Facilities account, same as for IDAaaS): [CatalogLogin](https://docs.mantidproject.org/algorithms/CatalogLogin-v1.html) and [CatalogGetDataFiles](https://docs.mantidproject.org/algorithms/CatalogGetDataFiles-v1.html)
- [ ] Saving/loading projects works 
 - [ ] Alter preferences in [File > Settings](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/useful/01_save_settings.html#settings) and check they are obeyed

rbauststfc, 2023-06-23:
There seems to be a new interface in the Interfaces->Direct list called Shiver. When I click on this I just see a warning "Shiver is not available".

SilkeSchomann, 2023-06-23:
Shiver needs to be pip installed before it can be used.

rbauststfc, 2023-06-23:
I just gave that a try (I closed Mantid, ran `pip install Shiver` in my conda environment and then restarted Mantid) and I still seem to get the same warning. If we're not concerned about this, though, then the rest of the GUIs under Interfaces opened OK.

rbauststfc, 2023-06-23:
Other than the issue with the Shiver interface, which I think we're not concerned about, the rest of this all looks good.

closed 2023-06-23 by rbauststfc