mantidproject/mantid #35750: Figure options in 3D surface plot not accepting colorbar limits

RichardWaiteSTFC, 2023-06-23:
**Describe the bug**
Figure options do not accept color axis limits (and possibly other settings)

**To Reproduce**
(1) Load a workspace
(2) Plot > 3D > Surface > Figure Options (cog) > image tab > change colorbar max limit

Gives red error message related to path at top of window
[image]
If you keep clicking OK it gets worse...
[image]

**Expected behavior**
No Error message, would accept input

jclarkeSTFC, 2023-10-17:
I couldn't reproduce this on Windows or Linux, was it happening with a particular file?

jclarkeSTFC, 2023-10-24:
I think this is the same problem as in #36264 and #36108, I can reproduce consistently using the instructions in #36264.

PR #36329 (2023-10-31): Fix crash in surface plot after changing properties
### Description of work
Sometimes surface plots were throwing exceptions after changing properties such as axis labels or colour bar limits. These changes stop that from happening and also fix another bug where the surface plot would keep shrinking every time you changed the limit on the colour bar.

#### Summary of work

- Removed call to `make_axes_area_auto_adjustable`. For unknown reasons this method (not one of ours) would cause an exception to be thrown (see issues referenced below) the next time `canvas.draw()` was called.
- Removed ar [... cut]
closed 2023-10-31 by SilkeSchomann