mantidproject/mantid #35828: Include More Metadata in ISIS SANS Output Files

ConorMFinn, 2023-07-20:
**Is your feature request related to a problem? Please describe.**
The instrument scientists would like to be able to see a wide range of additional information about the sample and the environment in their output files. This will reduce the amount of bookkeeping and cross-referencing required when looking at the output data. 

**Describe the solution you'd like**
As a ISIS SANS Interface User, I would like output from the interface to contain additional metadata, such as:
- [ ] Sample Environment State (Highest Priority)
  - [ ] Sample stack position (height, translation, angles, rotation)
  - [ ] Temperatures
  - [ ] Currents
  - [ ] Voltages
- [ ] Other Blocks tracked by IBEX

Development note: We need to consider how many of these data are loaded by mantid currently. Much of this information *should* be in the sample logs, but can often get lost if the load algorithm isn't written to keep them or explicitly load them. If it isn't already in the loaded workspaces, the first step would require adjustment of our loading algorithms. This would increase the care required when testing as it would effect all data being loaded into mantid.
- [ ] Investigate what metadata is currently already available in the loaded workspaces

**Describe alternatives you've considered**
- Can sample logs be copied directly from one workspace to another? Could we simply do this as a final step in the reduction?

**Additional context**
None yet.

smk78, 2023-11-13:
The issue of what metadata should be captured in _reduced_ data files arose in plenary discussions at the CanSAS-2023 workshop (Grenoble, 16-18 Oct 2023). There were passionate advocates of 'everything, everywhere, all at once'(!), but in reality some pragmatism will be necessary to keep things manageable.

Some thought will also need to be given to _how_ some of this additional metadata is stored, and _where_. For example, the full time-stamped SE log values or just the values at the start/end of that run, or something else, and stored in which file formats?

@dehoni has expressed an interest in having time-stamped SE log values so that the closest values can be associated with time-sliced reductions of an event dataset. Note that this would also lead to a new (maybe it already exists?) plotting requirement to enable, for example, I(Q) vs Q to be plotted against temperature or pressure or field strength, etc.

cailafinn, 2026-02-11:
Only because the Stale bot picked it up.