mantidproject/mantid #35835: Conda developer documentation

robertapplin, 2023-07-24:
**Description**
Add developer documentation that can be used to help understand how to use conda within Mantid, and useful tips for debugging issues. A couple of useful things to include could be:

- [ ] Using `conda search` command
- [ ] Marking a package as broken
- [ ] When should I consider pinning a package? Maybe we want a packaging pinning policy
- [ ] Use != when pinning rather than ==

robertapplin, 2023-07-24:
@thomashampson @sf1919  any other thoughts on what should be included?

MialLewis, 2023-08-21:
- Use of mamba
- Conda-tree
- How to launch local package builds for Linux/Windows/Mac
- Conda-forge feedstocks
- Creating environments (from file/from package/empty)
- Creating an environment in location (prefix)
- How and where packages are installed, and shared between environments.
- How conda packages are found by CMake?
- Overview of conda-build process build/host/run/test envs

robertapplin, 2023-09-21:
- Policy on use of pip in our conda environments

sf1919, 2024-03-21:
This still needs completing.