mantidproject/mantid #35847: Potential Issue with ViewBOA highlighted by assert statement.

jhaigh0, 2023-07-26:
**Describe the outcome that is desired.**
To look into the failing assert statement [here](https://github.com/mantidproject/mantid/blob/main/Framework/API/src/MatrixWorkspace.cpp#L636) when running `ViewBOA(2014, 61, 6)` in a debug build of Mantid.

Is this an underlying problem with the algorithm or an edge case not considered by the assert statement?

robertapplin, 2023-08-04:
Duplicate issue: https://github.com/mantidproject/mantid/issues/35888

peterfpeterson, 2023-08-04:
I need some help figuring out the right behavior. In "release" builds, the doctest that was copied into #35888 passes. The question is whether the number of input values should always equal the number of output values or whether the doctest has enshrined the wrong behavior.

Copying the "offending" code taken from [ViewBOA alg docs](https://docs.mantidproject.org/nightly/algorithms/ViewBOA-v1.html):
```py
import numpy as np

ViewBOA(2014, 61, 6)

lamdba_distribution = mtd['histogram61'].readY(0)
lambda_values = mtd['histogram61'].readX(0)

print("The wavelength spectrum is described by {} points.".format(len(lamdba_distribution)))
print("The range covered is {0} - {1} Angstrom.".format(round(lambda_values[0], 4),round(lambda_values[-1], 4)))
print("The maximum is located at {} Angstrom.".format(round(lambda_values[np.argmax(lamdba_distribution)], 4)))
```

PR #35890 (2023-08-07): Modify assert in MatrixWorkspace
The assert has been updated to match the code comments for the function.

The assert was originally introduced in #35800 which caused a bug in develop builds (assert is a no-op in release builds).

**To test:**

In a debug build (so assert is run) execute the python script
```py
ViewBOA(2014, 61, 6)
```

Fixes #35847

*This does not require release notes* because it was an issue that did not affect release builds.

---

#### Reviewer ####

Please comment on the points listed below ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html [... cut]
commit 8a6d6b27 (2023-08-07): Modify assert in MatrixWorkspace

The assert has been updated to match the code comments for the
function.

The assert was originally introduced in #35800 which caused a bug
in develop builds (assert is a no-op in release builds).

Fixes #35847
closed 2023-08-07 by SilkeSchomann