mantidproject/mantid #36004: Investigate `libopenblas 0.3.24` Causing Unit Tests Failures

ConorMFinn, 2023-09-05:
**Describe the outcome that is desired.**
Tests do not fail when this dependency is running this version. 

**Does the outcome relate directly to a problem? Please describe.**
Causing macOS unit test failures. 
- x86:
- ARM: 

**Describe any solutions you are considering**

**Additional context**

jclarkeSTFC, 2024-02-06:
I think we can close this one when #36790 is merged.

jclarkeSTFC, 2024-02-14:
Closing this now that #36790 is merged.

sf1919, 2024-08-20:
I'm reopening this one as the associated PR allowed v0.3.26 to be used but did not resolve the problem with v0.3.24 as this is still pinned. As v0.3.28 seems to have failures on similar tests we need a solution that will allow us to remove the v0.3.24 pin as well as v0.3.24 pin. See also #37834

closed 2026-02-10 by github-actions[bot]