mantidproject/mantid #36044: Manual Testing ORNL Powder Diffraction

robertapplin, 2023-09-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

See http://www.mantidproject.org/PowderDiffractionReduction

Kvieta1990, 2023-09-19:
Manual test was performed for the nightly version of Mantid on all the daily routines for powder diffraction at ORNL, including HFIR and SNS, and it seems to be working perfectly, except an issue for POWGEN GUI. We do have a ticket for it and hopefully it will be solved soon and will go into the release. Once that issue is solved, we should be all good.

robertapplin, 2023-09-19:
> Manual test was performed for the nightly version of Mantid on all the daily routines for powder diffraction at ORNL, including HFIR and SNS, and it seems to be working perfectly, except an issue [...]

Thanks for testing! Which ticket is the POWGEN GUI issue?

peterfpeterson, 2023-09-19:
Yuanpeng's issue is captured in #36118

Kvieta1990, 2023-09-20:
The issue with POWGEN GUI has been fixed with this PR, [https://github.com/mantidproject/mantid/pull/36135](https://github.com/mantidproject/mantid/pull/36135)

I grabbed the PR and tested it out and I can confirm that it is working perfectly now and producing consistent result as before.

In summary, I think we are good to proceed with the release on the ORNL powder diffraction relevant side.

Thank you!

closed 2023-10-18 by robertapplin