mantidproject/mantid #36045: Manual Testing ISIS Indirect interfaces

robertapplin, 2023-09-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

Follow the directions at: https://developer.mantidproject.org/Testing/IndirectInelastic/IndirectInelasticAcceptanceTests.html

jhaigh0, 2023-09-12:
Step 19 on test 2. The tab is now in the inelastic data manipulation interface

jhaigh0, 2023-09-12:
Could be useful to mention the isis sample data is needed at the top of the file

jhaigh0, 2023-09-12:
In Elwin tab, row of buttons has moved from bellow the plot to the bottom of the interface
[image]

jhaigh0, 2023-09-12:
Line 10 of Elwin test, should be vs Q2 not Q

jhaigh0, 2023-09-12:
F(Q) fit test, crash when selecting `EISFDiffSphereAlky` (regression)

MohamedAlmaki, 2023-09-12:
Reported by Spencer:

The save section in the ISIS Energy Transfer tab is always disabled, even after a successful run. After successfully running the tab, the section should be enabled.

jhaigh0, 2023-09-13:
Created issues
https://github.com/mantidproject/mantid/issues/36071
https://github.com/mantidproject/mantid/issues/36072
https://github.com/mantidproject/mantid/issues/36076
https://github.com/mantidproject/mantid/issues/36078

closed 2023-09-13 by jhaigh0