mantidproject/mantid #36047: Manual Testing Workbench Group 1- 4

robertapplin, 2023-09-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/Core/Core.html

thomashampson, 2023-09-12:
Regression on all OS:
No welcome message in messages widget

thomashampson, 2023-09-12:
on macOS: clicking question mark buttons opens help behind other windows (e.g. on plots, algorithm dialogs).

thomashampson, 2023-09-12:
all OS: in figure options, in the Axes tab enter an invalid latex string for the x label, e.g. Time ($\hellomu s$), and click apply. this is fine, it tells you what is wrong. Now go to Curves tab and remove the curve, there is an unhandled exception.

thomashampson, 2023-09-12:
superplot "normalise by max" button doesn't autoscale y-axis on clicking.

thomashampson, 2023-09-12:
home button on 3d plot doesn't reset to original orientation. Not sure if it should?

thomashampson, 2023-09-12:
Regression: editing axis label in 3D plot will cause unhandled exception.

thomashampson, 2023-09-12:
Regression:
in sliceviewer on macOS and Windows (Linux is fine), changing between min/max and other scale optionds will fluff up the autoscaling. Maybe introduced in this PR: #35857

thomashampson, 2023-09-12:
not regression:
double click on wireframe plot axis label and change label. When clicking OK, the plot thinks you're still grabbing it for a rotation.

thomashampson, 2023-09-12:
regression:
copy plot script to clipboard and paste into script editor, run the script, the plot is hidden by default. In 6.7 is showed by default.

thomashampson, 2023-09-12:
superplot "normalise by max" option isn't propagated to plot script generator.

MialLewis, 2023-09-13:
When showing detectors on a workspace with no instrument (such as an ASCII), the error message is handled but the message is unhelpful: `Error in execution of algorithm CreateDetectorTable: The instrument has no sample`. Prehaps in this case `Show Detectors` should be unselectable, as `Show Instrument`.

warunawickramasingha, 2023-09-13:
When attempting to add a user function to a fit, the button to open the function editor is difficult to capture since being very small.

MialLewis, 2023-09-15:
> Regression: in sliceviewer on macOS and Windows (Linux is fine), changing between min/max and other scale options will fluff up the autoscaling. Most likely introduced in this PR: #35857 edit: I can't reproduce this now...

I can reproduce this, but not with every dataset for some reason.

Issue: #36113

MialLewis, 2023-09-15:
All issues raised.

closed 2023-09-15 by MialLewis