mantidproject/mantid #36048: Manual Testing Muon

robertapplin, 2023-09-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

See instructions at: https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/index.html
https://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html

robertapplin, 2023-09-12:
The TF Asymmetry mode in this test needs to be changed at the top of the Fitting tab
https://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html#transverse-field-asymmetry-test

jhaigh0, 2023-09-12:
I ran into a problem in the second test where the run loading was failing, even attempting to load data for the first test (which had worked before) was failing. Mohammed got further but ran into this issue on the second to last test so we couldn't complete the tests.

robertapplin, 2023-09-13:
Issue to update the manual testing instructions: https://github.com/mantidproject/mantid/issues/36075

robertapplin, 2023-09-13:
> I ran into a problem in the second test where the run loading was failing, even attempting to load data for the first test (which had worked before) was failing. Mohammed got further but ran into this issue on the second to last test so we couldn't complete the tests.

I've opened an issue to investigate https://github.com/mantidproject/mantid/issues/36079 . But as we think its not a regression, its targeted at release 6.9

robertapplin, 2023-09-13:
All issues have been opened

closed 2023-09-13 by robertapplin