mantidproject/mantid #36050: Manual Testing FilterEvents

robertapplin, 2023-09-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://docs.mantidproject.org/interfaces/utility/Filter%20Events.html

rbauststfc, 2023-09-12:
When trying to enter 1.25e-7 into Filter by Log Min Value

Traceback (most recent call last):
  File "C:\MantidNightlyInstall\bin\lib\site-packages\mantidqtinterfaces\FilterEvents\eventFilterGUI.py", line 445, in set_minLogValue
    newminY = float(self.ui.lineEdit_5.text())
ValueError: could not convert string to float: '1.25e'

rbauststfc, 2023-09-12:
If using a TOF correction to sample (in advanced setup) that is not compatible with the workspace:

Error in execution of algorithm FilterEvents:
Inelastic instrument detector 0 of spectrum 0 does not have EFixed 
Traceback (most recent call last):
  File "C:\MantidNightlyInstall\bin\lib\site-packages\mantid\simpleapi.py", line 1069, in __call__
    algm.execute()
RuntimeError: Inelastic instrument detector 0 of spectrum 0 does not have EFixed 

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\MantidNightlyInstall\bin\lib\site-packages\mantidqtinterfaces\FilterEvents\eventFilterGUI.py", line 939, in filterByTime
    self.splitWksp(splitws, infows)
  File "C:\MantidNightlyInstall\bin\lib\site-packages\mantidqtinterfaces\FilterEvents\eventFilterGUI.py", line 1027, in splitWksp
    api.FilterEvents(
  File "C:\MantidNightlyInstall\bin\lib\site-packages\mantid\simpleapi.py", line 1076, in __call__
    raise RuntimeError(msg) from e
RuntimeError: FilterEvents-v1: Inelastic instrument detector 0 of spectrum 0 does not have EFixed

rbauststfc, 2023-09-12:
Testing instructions required.

rbauststfc, 2023-09-13:
Testing took approximately 20 minutes.

closed 2023-09-13 by rbauststfc