mantidproject/mantid #36052: Manual Testing Workspace Groups

robertapplin, 2023-09-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

* Follow the examples at http://docs.mantidproject.org/concepts/WorkspaceGroup.html 
* Ensure MantidWorkbench displays Group Workspaces correctly in the workspace tree

rbauststfc, 2023-09-12:
Examples in documentation should include all imports required - they seem to be relying on the imports that are automatically included in the Mantid editor to be there.

rbauststfc, 2023-09-13:
@robertapplin and @MialLewis, would it be worth reviewing the value of this manual test for the next release? It largely just involves running the documentation examples, and Richard made the good point that the doctests will be providing at least some coverage for this now. Perhaps it would be good to review what we really need to be manually checking for workspace groups for future releases.

closed 2023-09-13 by rbauststfc

robertapplin, 2023-09-13:
> @robertapplin and @MialLewis, would it be worth reviewing the value of this manual test for the next release? It largely just involves running the documentation examples, and Richard made the good [...]

That sounds like a good idea, can you create an issue to review this and assign it to myself and Mial?

rbauststfc, 2023-09-13:
I've opened issue #36077 for this.