mantidproject/mantid #36054: Manual Testing ISIS Reflectometry

robertapplin, 2023-09-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

http://developer.mantidproject.org/Testing/ReflectometryGUI/ReflectometryGUITests.html

gemmaguest, 2023-09-12:
In the [preview tab](https://developer.mantidproject.org/Testing/ReflectometryGUI/ReflectometryGUITests.html#preview-tab) section towards the end we could not select/edit some of the selection boxes on the sliceviewer plot - we could select the blue region and edit that ok, but not the pink or green regions. We had just previously experienced an error producing the 1D plot I think because we had originally selected an ROI without suitable data. Not sure if this was related to the subsequent problem being able to select regions.

gemmaguest, 2023-09-12:
We had a problem comparing `preview_reduced_ws` to `IvsQ_binned_45455` where the latter was not showing any data

SilkeSchomann, 2023-09-15:
I cannot reproduce any of the problems mentioned above.

closed 2023-09-15 by SilkeSchomann